mobizt/Firebase-ESP-Client #613: Firebase.ready() and <FirebaseData>.httpconnected() return true even if no internet is available

FedX33, 2023-12-18:
**Describe the bug**
When unplugging internet connection on router, while WiFi stays active, Firebase.ready() and <FirebaseData>.httpconnected() still return true. 
~4min after disconnecting internet 
message [E][WiFiClientImpl.h:347] tcpConnect(): connect on fd 48, errno: 118, "H
ost is unreachable" stops and insted
[E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app is printed every 30ms (This only happens if a stream with callback is used) 
83s later WiFi of ESP32 gets disconnected while router WiFi is still active

**To Reproduce**
unplug internet connection of router

**Expected behavior**
Fiebase.ready() and <FirebaseData>.httpconnected() shuld return false
[E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app should not be printed every 30ms

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [Core 6.1.11 Home 3.4.4]
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- Firebase Arduino Client Library for ESP8266 and ESP32 @ 4.4.9
|-- WiFi @ 2.0.0

Basic example with additional stream:

```
#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#elif __has_include(<WiFiNINA.h>)
#include <WiFiNINA.h>
#elif __has_include(<WiFi101.h>)
#include <WiFi101.h>
#elif __has_include(<WiFiS3.h>)
#include <WiFiS3.h>
#endif

#include <Firebase_ESP_Client.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "xxx"
#define WIFI_PASSWORD "xxx"
#define API_KEY "xxx"

#define DATABASE_URL "xxx"

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "xxx"
#define USER_PASSWORD "xxx"

// Define Firebase Data object
FirebaseData fbdo, stream;
bool dataChanged = false;
FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;
unsigned long checkConnectionPrevMillis = 0;
unsigned long count = 0;
[... 467 lines cut ...]
[377743][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
[377772][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
..
[380527][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
[380556][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
[380585][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
[380614][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
stream timed out, resuming... error reason: not connected[380644][E][WiFiGeneric
.cpp:1578] hostByName(): DNS Failed for terrarium-managment-system-default-rtdb.
europe-west1.firebasedatabase.app
[380673][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
..
[383631][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
stream timed out, resuming... error reason: not connected[383661][E][WiFiGeneric
.cpp:1578] hostByName(): DNS Failed for terrarium-managment-system-default-rtdb.
europe-west1.firebasedatabase.app
..
[386155][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
[386184][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
[386213][E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app
[386398] WiFi not connected
[396399] WiFi not connected

```
I deleted some prints because otherwise it would be 5605 lines

mobizt, 2023-12-18:
This is the normal as the `httpConnected` returns true since server connection established and returns false only when session was closed (by user) or network disconnected.

You have to use TCP keepalive for probing server connection periodically.
https://github.com/mobizt/Firebase-ESP-Client/blob/43d4a98621305899fc09936575f33eb01662c90e/examples/RTDB/DataChangesListener/Callback/Callback.ino#L163-L169

mobizt, 2023-12-18:
> Expected behavior
Fiebase.ready() and .httpconnected() shuld return false
[E][WiFiGeneric.cpp:1578] hostByName(): DNS Failed for test-default-rtdb.europe-west1.firebasedatabase.app should not be printed every 30ms'

Did you don't understand what I post before?  It is intended and normal behavior of this library.

You should set the ESP debug level in PlatformIO.

closed 2023-12-18 by mobizt

mobizt, 2023-12-18:
Firebase ready returns false only when network disconnected, or token expired or unauthenticated.