mobizt/Firebase-ESP-Client #623: token usage with Alexa skills or Google assistant

ateker01, 2024-01-13:
Greetings,

I apologize in advance that this is not in the scope of this very project but considering how helpful the followers of the library have been I was hoping someone will help me with this question.

I can read and write to the firebase using the tokens with email and password authentication. What I would like to achieve is to create Google Assistant and Alexa skill to read data from the firebase. However I can not figure out how to authenticate with email and password in other words the account linking part.
I have done extensive search with no luck so I wanted to reach out as I am hoping that someone also needed to setup a skill and they could direct me to a tutorial or a similar project page.

mobizt, 2024-01-13:
Sorry, this is the library repository issue section, and it is not community forum.

You should ask your question in community forum instead.

closed 2024-01-13 by mobizt

ateker01, 2024-01-13:
my bad! I will ask the question under the discussions