mobizt/Firebase-ESP-Client #631: Connection refused to firebase realtime database using my router working fine with my mobile data

lagmeshgavali12, 2024-01-26:
Describe the Bug:
When attempting to connect to Firebase Realtime Database (RTDB) using the provided code over Wi-Fi through a router, the connection is refused. However, the code works fine when using mobile data.

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <FirebaseESP8266.h>

// RTDB Helper for payload printing and other helper functions
#include <addons/RTDBHelper.h>

/* Define the WiFi credentials */
#define WIFI_SSID "YourWiFiSSID"
#define WIFI_PASSWORD "YourWiFiPassword"

/* Define the RTDB URL */
#define DATABASE_URL "YourFirebaseDatabaseURL"

/* Define the Firebase Data object */
FirebaseData fbdo;

/* Define the FirebaseAuth data for authentication data */
FirebaseAuth auth;

/* Define the FirebaseConfig data for config data */
FirebaseConfig config;

unsigned long dataMillis = 0;
int count = 0;

void setup() {
    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    // Configure Firebase
    config.database_url = DATABASE_URL;
    config.signer.test_mode = true;

    // Enable WiFi reconnection
    Firebase.reconnectNetwork(true);

    // Set the SSL buffer size
    fbdo.setBSSLBufferSize(4096, 1024);

    /* Initialize the library with Firebase auth and config */
    Firebase.begin(&config, &auth);
}

void loop() {
    if (millis() - dataMillis > 5000) {
        dataMillis = millis();
        Serial.printf("Set int... %s\n", Firebase.setInt(fbdo, "/test/int", count++) ? "ok" : fbdo.errorReason().c_str());
    }
}

[image]

IDE and Its Version:
Arduino IDE, PlatformIO
Version [2.2.1]

mobizt, 2024-01-26:
Your network problem not a library problem.

closed 2024-01-26 by mobizt

lagmeshgavali12, 2024-01-26:
Can you please help me with the next steps to establish reliable connectivity with Firebase and my ESP8266 device? I have checked my router, and the internet connectivity is fine. I am also able to open the Firebase website using the same router connection. The only issue is that I am unable to send or receive data via the ESP8266 controller.

mobizt, 2024-01-27:
It is your router problem.

You should ask this in community forum instead.

avinashboy, 2024-02-04:
Hi, it's me again. Could you send the community forum link?

mobizt, 2024-02-04:
@avinashboy 

You should search Google.