mobizt/Firebase-ESP-Client #633: The 'DISABLE_SD' Build-Flag does not work!

toqix, 2024-02-07:
**Describe the bug**
When using the 'DISABLE_SD' build-flag in PlatformIO the library still imports its SD libraries and tries to create its own SD objects. This leads to compile errors when working with a custom SD implementation such as SDFat on ESP32. 

**To Reproduce**
Using PlatformIO install the library and use any basic example, use the 'ESP32 Dev Kit' board, include "-D DISABLE_SD" in the platformio.ini file and try to compile the sketch.

**Expected behavior**
The build-flag should disable all SD implementations, thus no SD specific libraries or objects should be referenced or created.

**Screenshots**
When running a quick search through the library source code, you find that the only time "DISABLE_SD" is referenced is in the following comment: `Firebase Arduino Client Library for ESP8266 and ESP32/src/FirebaseFS.h:`
```
  158   * ⛔ Use following build flags to disable.
  159:  * -D DISABLE_SD or -DDISABLE_SD in PlatformIO
  160   */
```

**IDE and its version:**
 -  PlatformIO
 - Version [Latest]

**Board**
- ESP32 Dev Kit

**Additional context**
I will create a pull-request that tries to implement the expected behaviour soon.

mobizt, 2024-02-07:
That is the behavior of `PlatformIO` that will create all objects even that header was just included but the class will never be used.

You can compare the program size between disable and enable that option instead of just seeing compiling log.

closed 2024-02-07 by mobizt

toqix, 2024-02-07:
Removing 
```
#include <SD.h>
#define DEFAULT_SD_FS SD
#define CARD_TYPE_SD 1
```
from FirebaseFS.h definitely fixes the problem, however why shouldn't it be possible to add something like:
```
#if !defined(DISABLE_SD)

#include <SD.h>
#define DEFAULT_SD_FS SD
#define CARD_TYPE_SD 1

#endif
```
This should create the expected behaviour: not importing the SD library when the DISABLE_SD build-flag is used. Or am I wrong here?