mobizt/Firebase-ESP-Client #634: response payload read timed out

yousefmlotfy656, 2024-02-10:
hello dear , i found a problem with my code the after sometime i face a problems 
this is the serial monitor data
chstate is 1 [415482][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
[415484][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
response payload read timed out
[415500][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
response payload read timed out
[415512][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
response payload read timed out
[423483][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
[423485][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
response payload read timed out
[423501][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
response payload read timed out
[423513][E][WiFiClientImpl.h:468] tcpWrite(): fail on fd 48, errno: 128, "Socket is not connected"
response payload read timed out

i know that my internet connection not good enough but i want to make a stable code to reconnect again if any interuption happen in internet connection , so could you please help me in these part

this is my code , 
`/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
*/

// بتاريخ 1/6/2024
// moshklet el connection beyfsl ba3d 2 minutes stopped with line firebase ready
// but another problem exists is the retrieve the data from the firebase with brainstorm of doing a stream for the data from the firebase 
#include <Arduino.h>
#if defined(ESP32)
  #include <WiFi.h>
  #include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#elif defined(ESP8266)
  #include <ESP8266WiFi.h>
#endif
#include <Firebase_ESP_Client.h>

//Provide the token generation process info.
#include "addons/TokenHelper.h"
//Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

int timeout = 120; // seconds to run for
#define API_KEY "AIzaSyCiIe1cxteVJtZFQl-7BKXX_6PJWcxed5Y"

// Insert RTDB URLefine the RTDB URL */
#define DATABASE_URL "https://grad2023-bbcb3-default-rtdb.firebaseio.com/" 
#define Led_1 25
#define Led_2 26
#define Led_3 27
#define Led_4 14
#define Led_5 12
#define Led_6 4
#define Led_22 33
#define Input_1 34
#define Input_2 35
#define button1Pin 32
// Define Firebase Data object.
FirebaseData fbdo;

// Define firebase authentication.
FirebaseAuth auth;

// Definee firebase configuration.
FirebaseConfig config;
[... 266 lines cut ...]
        }
 }
      if (lichstate1 == 1) {
        Firebase.RTDB.setString(&fbdo, "/CarController/Light", light_State.c_str());
        Serial.print("Light_State changed to: ");
        Serial.println(light_State);
        lichstate1 = 0;

      }
      if(lichstate==1){
      Serial.print("lichstate is 1 ");
    if (light_State == "1") {
        digitalWrite(Led_3, HIGH);
        //Serial.println("LED 3 ON"); 
    }
    else {
        digitalWrite(Led_3, LOW);
  
       // Serial.println("LED 3 OFF");
    }
 lichstate=0;
      }
  }`

thank you dear.

mobizt, 2024-02-10:
After calling the RTDB function, if it gives the error, you can stop WiFiClient.

`fbdo.stopWiFiClient();`

I recommend you move to new library with simpler, faster and more reliable.
https://github.com/mobizt/FirebaseClient

closed 2024-02-10 by mobizt

yousefmlotfy656, 2024-02-10:
> After calling the RTDB function, if it gives the error, you can stop WiFiClient.
> 
> `fbdo.stopWiFiClient();`
> 
> I recommend you move to new library with simpler, faster and more reliable. https://github.com/mobizt/FirebaseClient

thanks for your response,

actually i didnt know where i should place the stop wifi client and do  you think this will solve the payload problem?

mobizt, 2024-02-10:
It is because session time out due to server terminates the connection.

You may have to stop `WiFiClient` when the RTDB function returns false at least every one minute to allow library to reconnect to the server as Google closed the session.

yousefmlotfy656, 2024-02-10:
> It is because session time out due to server terminates the connection.
> 
> You may have to stop `WiFiClient` when the RTDB function returns false at least every one minute to allow library to reconnect to the server as Google closed the session.

Thanks for your fast response ,
now i understand the problem but i didnt know what is the line of the rtdb function that returns false so please can you help me in this point (code)

thanks again.

mobizt, 2024-02-10:
The function where `fbdo` was used.

Anyway, as I suggested, you should consider or prepare to move to new library,

The migration is not too difficult, but much reduce your code complexity.

mobizt, 2024-02-10:
After you call to some function e.g. `Firebase.RTDB.setString(&fbdo`

If that function was called repeatedly for more than 1 minute, you may have to close the session (stop WiFiClient).

mobizt, 2024-02-10:
Your code was suitable for async operation or using stream instead.
https://github.com/mobizt/Firebase-ESP-Client/tree/main/examples/RTDB/DataChangesListener