mobizt/Firebase-ESP-Client #635: Firebase Storage Upload stops without any error message

JoergTiedemann, 2024-02-11:
Hi 
I use the lib in latest version 4.4.10 on esp32-cam and store jpg files to firebase storage with this function 
```
    if (Firebase.Storage.upload(&fbdo, STORAGE_BUCKET_ID /* Firebase Storage bucket id */, filename /* path to local file */, mem_storage_type_sd /* memory storage type, mem_storage_type_flash and mem_storage_type_sd */, BUCKET_PHOTO /* path of remote file stored in the bucket */, "image/jpeg" /* mime type */,fcsUploadCallback)){
      Serial.printf("\nDownload URL: %s\n", fbdo.downloadURL().c_str());
```
I use a sd-mmc card for local storage from where the picture was taken

Normally this work's fine and in the callback function I see the progress of upload
But sometimes the upload stops without any error message.
The FCS_UploadStatusInfo.Status has the value fb_esp_fcs_upload_status_error, but ErrorMsg is empty
What could be the reason?
Here the callback function 

```
void fcsUploadCallback(FCS_UploadStatusInfo info)
{
    if (info.status == fb_esp_fcs_upload_status_init){
        Serial.printf("Uploading file %s (%d) to %s\n", info.localFileName.c_str(), info.fileSize, info.remoteFileName.c_str());
    }
    else if (info.status == fb_esp_fcs_upload_status_upload)
    {
        Serial.printf("Uploaded %d%s, Elapsed time %d ms\n", (int)info.progress, "%", info.elapsedTime);
    }
    else if (info.status == fb_esp_fcs_upload_status_complete)
    {
        Serial.println("Upload completed\n");
        FileMetaInfo meta = fbdo.metaData();
        Serial.printf("Name: %s\n", meta.name.c_str());
        Serial.printf("Bucket: %s\n", meta.bucket.c_str());
        Serial.printf("contentType: %s\n", meta.contentType.c_str());
        Serial.printf("Size: %d\n", meta.size);
        Serial.printf("Generation: %lu\n", meta.generation);
        Serial.printf("Metageneration: %lu\n", meta.metageneration);
        Serial.printf("ETag: %s\n", meta.etag.c_str());
        Serial.printf("CRC32: %s\n", meta.crc32.c_str());
        Serial.printf("Tokens: %s\n", meta.downloadTokens.c_str());
        Serial.printf("Download URL: %s\n\n", fbdo.downloadURL().c_str());
        DiagManager.PushDiagData("Erledigt: %s",fbdo.downloadURL().c_str());
    }
    else if (info.status == fb_esp_fcs_upload_status_error){
        Serial.printf("Upload failed, %s\n", info.errorMsg.c_str());
    }
}
```
Thanks for any help in forward and kind regards
Joerg

mobizt, 2024-02-11:
It can be ESP32 core filesystem issue, you should use [SdFat](https://github.com/greiman/SdFat) library instead.

And update your board (ESP32/ESP8266) Arduino SDK.

JoergTiedemann, 2024-02-11:
Which version of core sdkmdid you recommend
2.0.14 ?

mobizt, 2024-02-11:
Why? Update means the latest in general as it is purpose of using more stable and lesser bugs.

JoergTiedemann, 2024-02-11:
Kook was just a question

closed 2024-02-11 by mobizt

JoergTiedemann, 2024-02-17:
Hi 
I've tried to use SDFat in firebase libray but no way
i got compilation error message:

`.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp: In member function 'uint8_t FB_RTDB::readQueueFileSdFat(FirebaseData*, SdFile&, QueueItem&, uint8_t)':
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp:1355:17: error: 'FBUtils::idle' is not a class member
         Core.ut.FBUtils::idle();
                 ^~~~~~~
.pio/libdeps/esp32dev/Firebase Arduino Client Library for ESP8266 and ESP32/src/rtdb/FB_RTDB.cpp:1362:29: error: 'FBUtils::idle' is not a class member
                     Core.ut.FBUtils::idle();
                             ^~~~~~~
*** [.pio\build\esp32dev\lib83a\Firebase Arduino Client Library for ESP8266 and ESP32\rtdb\FB_RTDB.cpp.o] Error 1
`
Iv added the folowing to CustomFirebaseFS.h

` #include <SdFat.h> //https://github.com/greiman/SdFat
#undef DEFAULT_SD_FS // remove SD defined macro
#undef CARD_TYPE_SD // remove SD defined macro

 static SdFat sd_fat_fs;   //should declare as static here
 #define DEFAULT_SD_FS sd_fat_fs
 #define CARD_TYPE_SD 1
 #define SD_FS_FILE SdFile`

The question is: what I have to do to integrate SdFat to firebase library ?