mobizt/Firebase-ESP-Client #639: SD cards only worjing with SD_MMC not SD Standard and not SDfat

JoergTiedemann, 2024-02-17:
I use the library on an ESP32-CAM board
If i use SD_MMC als SD filesystem it works till issue #635 
Then ir try sdfat and after fast fixing the issue #638 by your side i did'nt get any compiler errors
But still having trouble to fet sd file system to work
I use this code to upload:

` if (Firebase.Storage.upload(&fbdo, STORAGE_BUCKET_ID /* Firebase Storage bucket id */, filename /* path to local file */, mem_storage_type_sd /* memory storage type, mem_storage_type_flash and mem_storage_type_sd */, BUCKET_PHOTO /* path of remote file stored in the bucket */, "image/jpeg" /* mime type */,fcsUploadCallback)){
      Serial.printf("\nDownload URL: %s\n", fbdo.downloadURL().c_str());
    }
    else{
      Serial.println(fbdo.errorReason());
    }`
I got these error messages in the console:
`[ 36145][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
Upload failed, SD Storage is not ready`

For sure this is caused because of in the rest of the code I didn't mount the sd card. 
How did i have to mount or inititialize the sd card if i want to use SDfat special at the ESP32-CAM board 
I didn't get sdfat or standard sd into running with the library, just SD_MMC ist easy to mount and it will work

mobizt, 2024-02-17:
https://github.com/mobizt/Firebase-ESP-Client/blob/f17ea76ae870d6e02290504b1bd7e9599a3ca5cb/examples/Storage/FirebaseStorage/UploadFile/UploadFile.ino#L127

mobizt, 2024-02-17:
Please post in the discussion first.

closed 2024-02-17 by mobizt

mobizt, 2024-02-17:
You have to make your SD card to work first before use it with this library.

After it works you have to config the right SPI config (speed and Pins)
https://github.com/mobizt/Firebase-ESP-Client/blob/f17ea76ae870d6e02290504b1bd7e9599a3ca5cb/src/addons/SDHelper.h#L15-L19

And calling `SD_Card_Mounting()`; in your code which the following code in function `SD_Card_Mounting` will let the library knows your SD config.

https://github.com/mobizt/Firebase-ESP-Client/blob/f17ea76ae870d6e02290504b1bd7e9599a3ca5cb/src/addons/SDHelper.h#L66

mobizt, 2024-02-17:
If you ever seen the RTDB examples I have recommended before, you will see this comment which you have to follow.
https://github.com/mobizt/Firebase-ESP-Client/blob/f17ea76ae870d6e02290504b1bd7e9599a3ca5cb/examples/RTDB/File/SD/SD.ino#L14-L15

And this line to mount card.
https://github.com/mobizt/Firebase-ESP-Client/blob/f17ea76ae870d6e02290504b1bd7e9599a3ca5cb/examples/RTDB/File/SD/SD.ino#L134-L135

JoergTiedemann, 2024-02-18:
I've discussed with greiman at [https://github.com/greiman/SdFat/issues/469](url)
the correct initialisation on a ESP 32-CAM Board is
````
#include "SdFat.h"
SdFat SD;

#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13
SPIClass sdSPI(VSPI);
#define SD_CONFIG SdSpiConfig(SD_CS, USER_SPI_BEGIN | SHARED_SPI, SD_SCK_MHZ(10), &sdSPI)

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CONFIG)) { Serial.println("SD Card Mount Failed"); return; }
}
````
so if I change  SD_Card_Mountin() as following:
````
SPIClass spi;
// https://github.com/greiman/SdFat
SdSpiConfig sdFatSPIConfig(SPI_CS_PIN, USER_SPI_BEGIN | SHARED_SPI, SD_SCK_MHZ(SPI_CLOCK_IN_MHz),&spi);
bool SD_Card_Mounting()
{
    Serial.print("\nMounting SD Card... ");
    spi.begin(SPI_SCK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN, SPI_CS_PIN); // SPI pins config -> SCK,MISO, MOSI, SS
    if (!Firebase.sdBegin(&sdFatSPIConfig, SPI_CS_PIN, SPI_SCK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN)) // pointer to SdSpiConfig, SS, SCK,MISO, MOSI
    {
        Serial.println("failed\n");
        return false;
    }
    else
...
````
the system restarts at mounting
if i comment out the line spi.begin then already Firebase.sdBegin(&sdFatSPIConfig...
failed
so what I should have to do to intialize SDfat with firebase to get it running on a ESP32-CAM board ?