mobizt/Firebase-ESP-Client #641: Initializing of sdfat support at esp32-cam board

JoergTiedemann, 2024-02-18:
I've discussed with greiman at [https://github.com/greiman/SdFat/issues/469](url)
greiman says the correct initialisation f sdfat on a ESP 32-CAM Board is
````
#include "SdFat.h"
SdFat SD;

#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13
SPIClass sdSPI;
#define SD_CONFIG SdSpiConfig(SD_CS, USER_SPI_BEGIN | SHARED_SPI, SD_SCK_MHZ(10), &sdSPI)

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CONFIG)) { Serial.println("SD Card Mount Failed"); return; }
}
````
so if I change  SD_Card_Mounting() as following:
````
SPIClass spi;
// https://github.com/greiman/SdFat
SdSpiConfig sdFatSPIConfig(SPI_CS_PIN, USER_SPI_BEGIN | SHARED_SPI, SD_SCK_MHZ(SPI_CLOCK_IN_MHz),&spi);
bool SD_Card_Mounting()
{
    Serial.print("\nMounting SD Card... ");
    spi.begin(SPI_SCK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN, SPI_CS_PIN); // SPI pins config -> SCK,MISO, MOSI, SS
    if (!Firebase.sdBegin(&sdFatSPIConfig, SPI_CS_PIN, SPI_SCK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN)) // pointer to SdSpiConfig, SS, SCK,MISO, MOSI
    {
        Serial.println("failed\n");
        return false;
    }
    else
...
````
But the system restarts at mounting
if i comment out the line spi.begin to the following:
````
bool SD_Card_Mounting()
{
    Serial.print("\nMounting SD Card... ");
    //spi.begin(SPI_SCK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN, SPI_CS_PIN); // SPI pins config -> SCK,MISO, MOSI, SS
    if (!Firebase.sdBegin(&sdFatSPIConfig, SPI_CS_PIN, SPI_SCK_PIN, SPI_MISO_PIN, SPI_MOSI_PIN)) // pointer to SdSpiConfig, SS, SCK,MISO, MOSI
    {
        Serial.println("failed\n");
        return false;
    }
    else
...
````
SD_Cards_Mopunting already failed
so what I should have to do to intialize SDfat with firebase to get it running on a ESP32-CAM board ?

mobizt, 2024-02-18:
You should check your board for the available SPI ports that are not used for your board peripherals and flash chip.

Device crashed because of you used the SPI port that is not ready for use.

You should search Google for this.

closed 2024-02-18 by mobizt

JoergTiedemann, 2024-02-18:
you see in the working initialization in the code from greiman
this works with the standard examples from sdfat repo at my ESP32-CAM board:
````
#include "SdFat.h"
SdFat SD;

#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13
SPIClass sdSPI;
#define SD_CONFIG SdSpiConfig(SD_CS, USER_SPI_BEGIN | SHARED_SPI, SD_SCK_MHZ(10), &sdSPI)

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  sdSPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
  if (!SD.begin(SD_CONFIG)) { Serial.println("SD Card Mount Failed"); return; }
}
````
the only question is how can i do this initialization in firebase repo

mobizt, 2024-02-18:
I already posted in your previous issue and even state in the examples which you have to follow.

It is just to define the pins and SPI speed in SDHelper.h which will be used in the card mounting function.

You also have to define the required macros in the FSConfig.h which you are already done.
 
The method sdBegin in the card mounting function cannot cause your device to crash when you config the port and wiring correctly.

FYI. The compilation error is not related to the filesystem issue and it is because code clean up and optimization in latest version.

mobizt, 2024-02-18:
I never used that function and I think it is not related.

I hope that you should not call sdSPI.begin anywhere in your code as this will be done in the library internally to take control.

You just provide the correct setting of `sdFatSPIConfig` as in SDHelper.h which will pass as pointer to the function sdBegin.
`SdSpiConfig sdFatSPIConfig(SPI_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(SPI_CLOCK_IN_MHz));`

Note that the code in the SDHelper.h is just for working with examples to show how to initialize the filesystem via the sdBegin which is required to let the library know and take control.

You can define the static or global object of `sdFatSPIConfig` in your own sketch and calling `sdBegin` without including SDHelper.h in your code.