mobizt/Firebase-ESP-Client #642: tokein is not ready(revoked or expired) error

thinkupplus, 2024-02-19:
void init_Firebase()
{
  /* Assign the api key (required) */
  config.api_key = API_KEY;
  /* Assign the user sign in credentials */
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;
  /* Assign the RTDB URL */
  config.database_url = DATABASE_URL;
  // required for legacy HTTP API
  Firebase.FCM.setServerKey(FIREBASE_FCM_SERVER_KEY);
  fbdo.setResponseSize(2048);
  String base_path = "/default/";

  /* Assign the callback function for the long running token generation task */
  config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h
  /* Initialize the library with the Firebase authen and config */
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);

  if(Firebase.RTDB.setQueryIndex(&fbdo, "/default/" /* parent path of child's node that is being queried */, "" /* the child node key that is being queried */, DATABASE_SECRET))
  {
    _printf("Firebase is OK\n");
  }
  else 
  {
    1) error 
    _printf("error = %s\n", fbdo.errorReason().c_str());    
  }
  QueryFilter query;
  query.limitToLast(8);
  String s = "";
  if (Firebase.RTDB.getJSON(&fbdo, "/default", &query))
  {
    // printResult(fbdo.jsonString()); // see addons/RTDBHelper.h
    s = fbdo.jsonString();
  }

  _printf("firebase result = %s\n", s);

  // Clear all query parameters
  query.clear();

  StaticJsonDocument<CHARLENGTH> doc;
  DeserializationError error = deserializeJson(doc, s);

  if (error)
  {
    2) error 
    _printf("deserializeJson() failed:  %s\n", error.c_str());    

  }

  String s1 = doc["mqtt_uri"];
  String s2 = doc["firmware"]["mopeka"]["version"];
  String szFBURI = doc["firmware"]["mopeka"]["path"];

  _printf("mqtt_url = %s,  szFBVersion = %s, url = %s\n", s1.c_str(), s2.c_str(), szFBURI.c_str());

  if (s1.equals("null"))
  {
    s1 = "ucore.iptime.org";
  }

  if (s2.equals("null"))
  {
    s2 = "7";
  }

  if (szFBURI.equals("null"))
  {
    szFBURI = "firmware/mopeka/firmware.bin";
  }

  network.szMqttServer = s1;

  eprom.writeStringToFlash(s1.c_str(), ER_MQTTSERVER);
  doc.clear();
}

**IDE and its version:**
 - Arduino, PlatformIO
 - Version [e.g. 1.8.9]

; platform = espressif32 ;https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/829/framework-arduinoespressif32-v4.4.1-3010c4571f.tar.gz
board = esp32-s3-devkitc-1
framework = arduino
board_build.mcu = esp32
;upload_speed = 921600
;upload_port = COM10
monitor_speed = 115200
board_upload.flash_size = 8MB
board_build.partitions=default_8MB.csv    
build_flags = 
	-O2 
    -ffunction-sections 
    -fdata-sections 
    -Wl,--gc-sections 
lib_deps = 
	bblanchon/ArduinoJson@^6.20.0
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit INA219@^1.2.1
	mobizt/Firebase Arduino Client Library for ESP8266 and ESP32@^4.4.10

pio error message log
1) error = token is not ready (revoked or expired)
2) deserializeJson() failed:  EmptyInput

This is a source that retrieves data using Firebase's realtime database and connects to the MQTT server.
This is the source code that has been widely used in the meantime.
But at some point it didn't work.
The error is that token is not ready, but the DB access permission is set to true.
I searched for the same phenomenon for several days and tried entering admin tokens, but the error is the same.
Is it because Firebase's policy changed? I am leaving a message because I cannot solve the problem on my own.

mobizt, 2024-02-19:
You should follow the library examples for correct usage.
https://github.com/mobizt/Firebase-ESP-Client/blob/f17ea76ae870d6e02290504b1bd7e9599a3ca5cb/examples/RTDB/Basic/Basic.ino#L176-L181

closed 2024-02-19 by mobizt

thinkupplus, 2024-02-20:
/**
 * Created by K. Suwatchai (Mobizt)
 *
 * Email: k_suwatchai@hotmail.com
 *
 * Github: https://github.com/mobizt/FirebaseJson
 *
 * Copyright (c) 2023 mobizt
 *
 */

#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif

#include <Firebase_ESP_Client.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID "******"
#define WIFI_PASSWORD "******"

// For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

/* 2. Define the API Key */
#define API_KEY "AIz******H2jXuB24bQwjfw"

/* 3. Define the RTDB URL */
#define DATABASE_URL "https://sma******t-rtdb.firebaseio.com" //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "******"
#define USER_PASSWORD "******"

// Define Firebase Data object
FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

unsigned long count = 0;

void setup()
{

    Serial.begin(115200);

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    Serial.print("Connecting to Wi-Fi");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    /* Assign the api key (required) */
    config.api_key = API_KEY;

    /* Assign the user sign in credentials */
    auth.user.email = USER_EMAIL;
    auth.user.password = USER_PASSWORD;

    /* Assign the RTDB URL (required) */
    config.database_url = DATABASE_URL;

    /* Assign the callback function for the long running token generation task */
    config.token_status_callback = tokenStatusCallback; // see addons/TokenHelper.h

    // Since Firebase v4.4.x, BearSSL engine was used, the SSL buffer need to be set.
    // Large data transmission may require larger RX buffer, otherwise connection issue or data read time out can be occurred.
    fbdo.setBSSLBufferSize(2048 /* Rx buffer size in bytes from 512 - 16384 */, 1024 /* Tx buffer size in bytes from 512 - 16384 */);

    // Or use legacy authenticate method
    // config.database_url = DATABASE_URL;
    // config.signer.tokens.legacy_token = "<database secret>";

    Firebase.begin(&config, &auth);

    // Comment or pass false value when WiFi reconnection will control by your code or third party library e.g. WiFiManager
    Firebase.reconnectNetwork(true);
}

void loop()
{
    // Flash string (PROGMEM and  (FPSTR), String C/C++ string, const char, char array, string literal are supported
    // in all Firebase and FirebaseJson functions, unless F() macro is not supported.

    // Firebase.ready() should be called repeatedly to handle authentication tasks.

    if (Firebase.ready() && (millis() - sendDataPrevMillis > 15000 || sendDataPrevMillis == 0))
    {
        sendDataPrevMillis = millis();

        FirebaseJson json;
        json.setDoubleDigits(3);
        json.add("value", count);

        Serial.printf("Set json... %s\n", Firebase.RTDB.setJSON(&fbdo, "/test/json", &json) ? "ok" : fbdo.errorReason().c_str());

        Serial.printf("Get json... %s\n", Firebase.RTDB.getJSON(&fbdo, "/test/json") ? fbdo.to<FirebaseJson>().raw() : fbdo.errorReason().c_str());

        FirebaseJson jVal;
        Serial.printf("Get json ref... %s\n", Firebase.RTDB.getJSON(&fbdo, "/test/json", &jVal) ? jVal.raw() : fbdo.errorReason().c_str());

        FirebaseJsonArray arr;
        arr.setFloatDigits(2);
        arr.setDoubleDigits(4);
        arr.add("a", "b", "c", true, 45, (float)6.1432, 123.45692789);

        Serial.printf("Set array... %s\n", Firebase.RTDB.setArray(&fbdo, "/test/array", &arr) ? "ok" : fbdo.errorReason().c_str());

        Serial.printf("Get array... %s\n", Firebase.RTDB.getArray(&fbdo, "/test/array") ? fbdo.to<FirebaseJsonArray>().raw() : fbdo.errorReason().c_str());

        Serial.printf("Push json... %s\n", Firebase.RTDB.pushJSON(&fbdo, "/test/push", &json) ? "ok" : fbdo.errorReason().c_str());

        // json.set("value", count + 0.29745);
        // Serial.printf("Update json... %s\n\n", Firebase.RTDB.updateNode(&fbdo, "/test/push/" + fbdo.pushName(), &json) ? "ok" : fbdo.errorReason().c_str());

        count++;
    }
}

Even though I just copied and pasted the example provided in the sample. The error details are the same.
The rules of Firebase's realtime database are .read: true, .write: "auth.uid != null".

I've been suffering from the same error for several days.

console  log
Connecting to Wi-Fi....
Connected with IP: 192.168.0.20

Firebase Client v4.4.11

Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = ready
Set json... token is not ready (revoked or expired)
Get json... token is not ready (revoked or expired)
Get json ref... token is not ready (revoked or expired)
Set array... token is not ready (revoked or expired)
Get array... token is not ready (revoked or expired)
Push json... token is not ready (revoked or expired)

mobizt, 2024-02-20:
The problem is on your side which you should be solved. 

The library works normally and test 24/7 for years.

You should try with clean library installation and new Firebase project.

ateker01, 2024-03-15:
Hello @thinkupplus, 
Have you ever figured this out?
Interestingly I can set but can not get

14:54:26.000 -> Token info: type = id token (GITKit token), status = on request
14:54:27.242 -> Token info: type = id token (GITKit token), status = ready
14:54:29.336 -> Set Wifi connected success
14:54:29.602 -> Set epoch success
14:54:30.630 -> Get int... token is not ready (revoked or expired)

I made sure that the database rules are not the reason

mobizt, 2024-03-16:
@ateker01 

If the time was changed or set after authenticated, you have to call `Firebse.begin` again to re-authenticate.

thinkupplus, 2024-03-19:
> Hello @thinkupplus, Have you ever figured this out? Interestingly I can set but can not get
>
> 14:54:26.000 -> Token info: type = id token (GITKit token), status = on request 14:54:27.242 -> Token [...]

I haven't solved this problem yet

ateker01, 2024-03-19:
I had actually have multiple issue and I looked at a little deeper after @mobizt's comment. 

I was calling the firebase authentication too often. Once I removed that, the token failure stopped.

mobizt, 2024-03-19:
@thinkupplus 

You should try to use the latest sdk and platform by ignoring these specific.
```
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.3rc1/platform-espressif32-2.0.3new.zip
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/829/framework-arduinoespressif32-v4.4.1-3010c4571f.tar.gz
```

It should be ok if you use the mainstream platform and packages.