mobizt/Firebase-ESP-Client #643: Firebase Storage Upload stops without any error message also if using sdfat as sd filesystem

JoergTiedemann, 2024-02-19:
Hi 
I use the lib in latest version 4.4.11 on esp32-cam and store jpg files to firebase storage with this function 
```
    if (Firebase.Storage.upload(&fbdo, STORAGE_BUCKET_ID /* Firebase Storage bucket id */, filename /* path to local file */, mem_storage_type_sd /* memory storage type, mem_storage_type_flash and mem_storage_type_sd */, BUCKET_PHOTO /* path of remote file stored in the bucket */, "image/jpeg" /* mime type */,fcsUploadCallback)){
      Serial.printf("\nDownload URL: %s\n", fbdo.downloadURL().c_str());
```
I use a sdfat as a sd filesystem for local storage from where the picture was taken and adruino cor is also updated to latested version

Normally this work's fine and in the callback function I see the progress of upload
But sometimes the upload stops without any error message.
The FCS_UploadStatusInfo.Status has the value fb_esp_fcs_upload_status_error, but ErrorMsg is empty
What could be the reason?
Here the callback function 
```
void fcsUploadCallback(FCS_UploadStatusInfo info)
{
    if (info.status == fb_esp_fcs_upload_status_init){
        Serial.printf("Uploading file %s (%d) to %s\n", info.localFileName.c_str(), info.fileSize, info.remoteFileName.c_str());
    }
    else if (info.status == fb_esp_fcs_upload_status_upload)
    {
        Serial.printf("Uploaded %d%s, Elapsed time %d ms\n", (int)info.progress, "%", info.elapsedTime);
    }
    else if (info.status == fb_esp_fcs_upload_status_complete)
    {
        Serial.println("Upload completed\n");
        FileMetaInfo meta = fbdo.metaData();
        Serial.printf("Name: %s\n", meta.name.c_str());
        Serial.printf("Bucket: %s\n", meta.bucket.c_str());
        Serial.printf("contentType: %s\n", meta.contentType.c_str());
        Serial.printf("Size: %d\n", meta.size);
        Serial.printf("Generation: %lu\n", meta.generation);
        Serial.printf("Metageneration: %lu\n", meta.metageneration);
        Serial.printf("ETag: %s\n", meta.etag.c_str());
        Serial.printf("CRC32: %s\n", meta.crc32.c_str());
        Serial.printf("Tokens: %s\n", meta.downloadTokens.c_str());
        Serial.printf("Download URL: %s\n\n", fbdo.downloadURL().c_str());
        DiagManager.PushDiagData("Erledigt: %s",fbdo.downloadURL().c_str());
    }
    else if (info.status == fb_esp_fcs_upload_status_error){
        Serial.printf("Upload failed, %s\n", info.errorMsg.c_str());
    }
}
```
so it is the same behavior as described in  #issue#635
but for now I#ve sd filesystem woking (10Mhz speed) and also latest version of ESP32 Adruino core sdk
but it is already the same just stopping without any message :
````
Uploading picture... Uploading file /picture_2024-02-19_18-11-15.jpg (160696) to /data/photo.jpg
Uploaded 0%, Elapsed time 91839 ms
Uploaded 2%, Elapsed time 91863 ms
Uploaded 4%, Elapsed time 91891 ms
Uploaded 6%, Elapsed time 91924 ms
Uploaded 8%, Elapsed time 91955 ms
Uploaded 10%, Elapsed time 91984 ms
Uploaded 12%, Elapsed time 92013 ms
Uploaded 14%, Elapsed time 92037 ms
Uploaded 16%, Elapsed time 92070 ms
Uploaded 18%, Elapsed time 92095 ms
Upload failed, 
````
if i try again it just works
Thanks for any help in forward and kind regards

mobizt, 2024-02-19:
As I ever mentioned, you should use discussion first.

Your problem is not library issue.

The SPI speed may be too fast that causes data error.

closed 2024-02-19 by mobizt

JoergTiedemann, 2024-02-19:
Ok will slow down spi speed to 2MHz and try again

JoergTiedemann, 2024-02-24:
Hi mobizt
so i've made many test with different spi speeds
if I slow down SPI speed down to 2MHz, 1 MHZ, 0.5MHZ or down to 0.1 MHZ it changed nothing
most times it runs but the I've several times the upload stops at different percentage sometimes at 18% sometimes 25% sometimes 60-70%
the prohability the upload stops does not regading to the SPI speed, the problem occurs often (1 stop every 10 uploads), doesn't matter of the SPI speed
I've to try again a few secondes or a few minutes later then it works
I don't think that the problem is caused in the spi speed it seems the problem is caused by a different reason. How can i detect the reason the upload failed ?
Erroor message would be really helpful

mobizt, 2024-02-24:
I have no idea and you have to debug it yourself. 
The SdFat and Firebase library are not the issue, and it is stable enough from intensively testing.

In addition to hardware issue, you have to manage your code to minimize the stack usage by avoiding using service routine of interrupt and callback.

JoergTiedemann, 2024-02-24:
Yes I've checked the code according to stack and memory usage
I will try to debug but where in your code will the failure to stop upload and call to the callback routine be generated ?
The problem is the upload code detects a problem and stop upload but didn't tell me the reason

mobizt, 2024-02-24:
I understand and what I have said, you have to debug it yourself and as I always said that I test this intensively 24/7 and everything works fine.

You already know that the filesystem is not part of library and it implemented via fs class. 

Any read and write failures cause upload and download stop with unknown reason and cannot fix instead of network data transmission issue that we can handle.

mobizt, 2024-02-24:
I also don't have any issue in ESP8266 with has limited memory and RaspberryPi Pico for filesystem test.

JoergTiedemann, 2024-02-24:
yes i understand what you said that i've to debug myself
I will try to debug but where in your code will the failure to stop upload and call to the callback routine be generated ?
Just for helping me where i have to go my frist step
It's really a lot of stuff and difficult to understand, i just want to know where (in which file / function) you generate the upload abort and setting the FCS_UploadStatusInfo.status to fb_esp_fcs_upload_status_error
and where then you call the upload callbackroutine
I don't say that there is any failure in your code where you are responsible. I just want to find out whats going wrong and what causes the abort and asking for some help

mobizt, 2024-02-24:
There is none of my statement that I think you blame the library I just confirm it that is no bug in the library.

JoergTiedemann, 2024-02-24:
Ok no problem
I will try to debug but where in your code will in case of a failure to stop upload and to call the callback routine be generated ?
Just for helping me where i have to go my frist step
I just want to know where (in which file / function) you generate the upload abort and setting the FCS_UploadStatusInfo.status to fb_esp_fcs_upload_status_error
and where then you call the upload callbackroutine ?

mobizt, 2024-02-24:
It is up to you, and it is wasted the time and seems to be useless to debug in library code.

Here is the simple test code.

```cpp
#include <Arduino.h>
#include <WiFi.h>

#include <Firebase_ESP_Client.h>
#include <addons/TokenHelper.h>
#include <addons/SDHelper.h>
#define WIFI_SSID "WIFI_AP"
#define WIFI_PASSWORD "WIFI_PASSWORD"
#define API_KEY "API_KEY"
#define USER_EMAIL "USER_EMAIL"
#define USER_PASSWORD "USER_PASSWORD"
#define STORAGE_BUCKET_ID "BUCKET-NAME.appspot.com"

FirebaseData fbdo;

FirebaseAuth auth;
FirebaseConfig config;

bool taskCompleted = false;

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
WiFiMulti multi;
#endif

unsigned long ms = 0;
int count = 0;

void setup()
{

    Serial.begin(115200);
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

    Serial.print("Connecting to Wi-Fi");
    unsigned long ms = millis();
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    configTime(7 * 3600, 0, "pool.ntp.org");

    while (time(nullptr) < 1708775869)
    {
        delay(0);
    }

    config.api_key = API_KEY;
    auth.user.email = USER_EMAIL;
    auth.user.password = USER_PASSWORD;

    config.token_status_callback = tokenStatusCallback;

    Firebase.reconnectNetwork(true);

    fbdo.setBSSLBufferSize(4096, 1024);

    Firebase.begin(&config, &auth);

    config.fcs.upload_buffer_size = 512;

    SD_Card_Mounting();
}

void fcsUploadCallback(FCS_UploadStatusInfo info)
{
    if (info.status == firebase_fcs_upload_status_init)
    {
        Serial.printf("Uploading file %s (%d) to %s\n", info.localFileName.c_str(), info.fileSize, info.remoteFileName.c_str());
    }
    else if (info.status == firebase_fcs_upload_status_upload)
    {
        Serial.printf("Uploaded %d%s, Elapsed time %d ms\n", (int)info.progress, "%", info.elapsedTime);
    }
    else if (info.status == firebase_fcs_upload_status_complete)
    {
        Serial.println("Upload completed\n");
        FileMetaInfo meta = fbdo.metaData();
        Serial.printf("Name: %s\n", meta.name.c_str());
        Serial.printf("Bucket: %s\n", meta.bucket.c_str());
        Serial.printf("contentType: %s\n", meta.contentType.c_str());
        Serial.printf("Size: %d\n", meta.size);
        Serial.printf("Generation: %lu\n", meta.generation);
        Serial.printf("Metageneration: %lu\n", meta.metageneration);
        Serial.printf("ETag: %s\n", meta.etag.c_str());
        Serial.printf("CRC32: %s\n", meta.crc32.c_str());
        Serial.printf("Tokens: %s\n", meta.downloadTokens.c_str());
        Serial.printf("Download URL: %s\n\n", fbdo.downloadURL().c_str());
    }
    else if (info.status == firebase_fcs_upload_status_error)
    {
        Serial.printf("Upload failed, %s\n", info.errorMsg.c_str());
    }
}

void loop()
{

    if (Firebase.ready() && millis() - ms > 10000)
    {
        ms = millis();
        count++;
        Serial.println("\nUpload file...\n");
        Serial.printf("run: %d, timestamp: %d\n", count, time(nullptr));

        if (!Firebase.Storage.upload(&fbdo, STORAGE_BUCKET_ID, "/media.mp4", mem_storage_type_sd, "media.mp4", "video/mp4", fcsUploadCallback))
            Serial.println(fbdo.errorReason());
    }
}

```

mobizt, 2024-02-24:
Here is the 100 runs upload result of 383k media file.
And everything is ok.

```yaml
Connecting to Wi-Fi..............
Connected with IP: 192.168.137.118

Firebase Client v4.4.11

Token info: type = id token (GITKit token), status = on request
Token info: type = id token (GITKit token), status = ready

Mounting SD Card... success

Upload file...

run: 1, timestamp: 1708776007
Uploading file /media.mp4 (383631) to media.mp4
Uploaded 0%, Elapsed time 11602 ms
Uploaded 2%, Elapsed time 12225 ms
Uploaded 4%, Elapsed time 13129 ms
Uploaded 6%, Elapsed time 13748 ms
Uploaded 8%, Elapsed time 14665 ms
Uploaded 10%, Elapsed time 15285 ms
Uploaded 12%, Elapsed time 16201 ms
Uploaded 14%, Elapsed time 16820 ms
Uploaded 16%, Elapsed time 17737 ms
Uploaded 18%, Elapsed time 18356 ms
Uploaded 20%, Elapsed time 19274 ms
Uploaded 22%, Elapsed time 19892 ms
Uploaded 24%, Elapsed time 20809 ms
Uploaded 26%, Elapsed time 21428 ms
Uploaded 28%, Elapsed time 22345 ms
Uploaded 30%, Elapsed time 22964 ms
Uploaded 32%, Elapsed time 23881 ms
Uploaded 34%, Elapsed time 24501 ms
Uploaded 36%, Elapsed time 25417 ms
Uploaded 38%, Elapsed time 26036 ms
Uploaded 40%, Elapsed time 26953 ms
Uploaded 42%, Elapsed time 27572 ms
Uploaded 44%, Elapsed time 28489 ms
Uploaded 46%, Elapsed time 29108 ms
Uploaded 48%, Elapsed time 30025 ms
Uploaded 50%, Elapsed time 30644 ms
Uploaded 52%, Elapsed time 31564 ms
Uploaded 54%, Elapsed time 32180 ms
Uploaded 56%, Elapsed time 33097 ms
Uploaded 58%, Elapsed time 33716 ms
Uploaded 60%, Elapsed time 34633 ms
Uploaded 62%, Elapsed time 35252 ms
Uploaded 64%, Elapsed time 36169 ms
Uploaded 66%, Elapsed time 36788 ms
Uploaded 68%, Elapsed time 37707 ms
Uploaded 70%, Elapsed time 38324 ms
Uploaded 72%, Elapsed time 39242 ms
Uploaded 74%, Elapsed time 39860 ms
Uploaded 76%, Elapsed time 40777 ms
Uploaded 78%, Elapsed time 41396 ms
Uploaded 80%, Elapsed time 42313 ms
Uploaded 82%, Elapsed time 42933 ms
Uploaded 84%, Elapsed time 43849 ms
Uploaded 86%, Elapsed time 44468 ms
Uploaded 88%, Elapsed time 45385 ms
Uploaded 90%, Elapsed time 46004 ms
Uploaded 92%, Elapsed time 46921 ms
Uploaded 94%, Elapsed time 47543 ms
Uploaded 96%, Elapsed time 48457 ms
Uploaded 98%, Elapsed time 49076 ms
Uploaded 100%, Elapsed time 49694 ms
Upload completed
[... 5437 lines cut ...]
Uploaded 98%, Elapsed time 3250994 ms
Uploaded 100%, Elapsed time 3251613 ms
Upload completed

Name: media.mp4
Bucket: xxxxxxxxxxxxx.appspot.com
contentType: video/mp4
Size: 383631
Generation: 1708779249
Metageneration: 1
ETag: CLaaxOGCxIQDEAE=
CRC32: lo9D5w==
Tokens: fcfb2ed3-90b9-491f-8165-74baa123ed15
Download URL: https://firebasestorage.googleapis.com/v0/b/xxxxxxxxxxxxx.appspot.com/o/media.mp4?alt=media&token=fcfb2ed3-90b9-491f-8165-74baa123ed15

Upload file...

run: 82, timestamp: 1708779249
Uploading file /media.mp4 (383631) to media.mp4
Uploaded 0%, Elapsed time 3253457 ms
Uploaded 2%, Elapsed time 3254066 ms
Uploaded 4%, Elapsed time 3254984 ms
Uploaded 6%, Elapsed time 3255602 ms
Uploaded 8%, Elapsed time 3256521 ms
Uploaded 10%, Elapsed time 3257138 ms

JoergTiedemann, 2024-02-24:
Ok thank you for example testcode
I will try to debug in library code to find out the reason for upload abort (also if it's wasting time because of reason is for sure not the library ;)
where in your code will in case of a failure to stop upload and to call the callback routine be generated ?
Just for helping me
I just want to know where (in which file / function) you generate the upload abort and setting the FCS_UploadStatusInfo.status to fb_esp_fcs_upload_status_error
and where then you call the upload callbackroutine ?

mobizt, 2024-02-24:
It is useless because hardware is defected.