mobizt/Firebase-ESP-Client #651: Unknown Parentpath data

lchvsanilkumar, 2024-03-10:
`/**
 * Created by K. Suwatchai (Mobizt)
 *
 * Email: k_suwatchai@hotmail.com
 *
 * Github: https://github.com/mobizt/Firebase-ESP-Client
 *
 * Copyright (c) 2023 mobizt
 *
 */

#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#elif __has_include(<WiFiNINA.h>)
#include <WiFiNINA.h>
#elif __has_include(<WiFi101.h>)
#include <WiFi101.h>
#elif __has_include(<WiFiS3.h>)
#include <WiFiS3.h>
#endif

#include <Firebase_ESP_Client.h>

// Provide the token generation process info.
#include <addons/TokenHelper.h>

// Provide the RTDB payload printing info and other helper functions.
#include <addons/RTDBHelper.h>

/* 1. Define the WiFi credentials */
#define WIFI_SSID ""
#define WIFI_PASSWORD ""

// For the following credentials, see examples/Authentications/SignInAsUser/EmailPassword/EmailPassword.ino

/* 2. Define the API Key */
#define API_KEY ""

/* 3. Define the RTDB URL */
#define DATABASE_URL "testing-912c2.firebaseio.com"  //<databaseName>.firebaseio.com or <databaseName>.<region>.firebasedatabase.app

/* 4. Define the user Email and password that alreadey registerd or added in your project */
#define USER_EMAIL "@gmail.com"
#define USER_PASSWORD "123456789"

// Define Firebase Data object
FirebaseData fbdo;
FirebaseData fbdo1;
FirebaseData fbdo2;
FirebaseData fbdo3;

FirebaseData stream;

FirebaseAuth auth;
FirebaseConfig config;

unsigned long sendDataPrevMillis = 0;

String parentPaths[] = { "/AK401", "/AK402", "/AK403" };
String childPath[] = { "/S1", "/S2", "/S3", "/S4", "/S4D", "/S5D", "/Triggers/11", "/Triggers/12", "/Triggers/21", "/Triggers/22", "/Triggers/31", "/Triggers/32", "/Triggers/41", "/Triggers/42", "/Triggers/51", "/Triggers/52", "/Triggers/61", "/Triggers/62", "/Triggers/hour11", "/Triggers/hour12", "/Triggers/hour21", "/Triggers/hour22", "/Triggers/hour31", "/Triggers/hour32", "/Triggers/hour41", "/Triggers/hour42", "/Triggers/hour51", "/Triggers/hour52", "/Triggers/hour61", "/Triggers/hour62", "/Triggers/min11", "/Triggers/min12", "/Triggers/min21", "/Triggers/min22", "/Triggers/min31", "/Triggers/min32", "/Triggers/min41", "/Triggers/min42", "/Triggers/min51", "/Triggers/min52", "/Triggers/min61", "/Triggers/min62", "/Triggers/countback", "/Triggers/refreshvalue", "/Triggers/restart" };

int count = 0;

volatile bool dataChanged = false;

#if defined(ARDUINO_RASPBERRY_PI_PICO_W)
WiFiMulti multi;
[... 177 lines cut ...]

    Serial.println("ok\n");
  }

  if (dataChanged) {
    dataChanged = false;
    // When stream data is available, do anything here...
  }

  // After calling stream.keepAlive, now we can track the server connecting status
  if (!stream.httpConnected()) {
    // Server was disconnected!
  }
}

// To pause stream
// stream.pauseFirebase(true);
// stream.clear(); // close session and release memory

// To resume stream with callback
// stream.pauseFirebase(false);
// Firebase.RTDB.setMultiPathStreamCallback(&stream, streamCallback, streamTimeoutCallback);`

i am gettting the data but the data is coming from which parentpath is unknown.        Serial.printf("path: %s, event: %s, type: %s, value: %s%s", stream.dataPath.c_str(), stream.eventType.c_str(), stream.type.c_str(), stream.value.c_str(), i < numChild - 1 ? "\n" : "");
 this gives the child datapath, but i want along with parent path...is it possible....please help me

mobizt, 2024-03-10:
The absolute path is parent + child.

closed 2024-03-10 by mobizt

lchvsanilkumar, 2024-03-10:
stream.dataPath.c_str() here it is child node path but i want along the parentpath because here i am using multiple parent nodes

mobizt, 2024-03-10:
You should separate the callback for each stream if you don't know how to get the parent path.
It is your code design, and you should solve it yourself.

lchvsanilkumar, 2024-03-10:
thank you for your huge help