mobizt/Firebase-ESP-Client #653: ESP32 Becomes Unresponsive After 12 Hours

Saipreetham0, 2024-03-11:
After approximately 12 hours of operation, the ESP32 microcontroller ceases to respond. Symptoms include a blank LCD screen, indicating a lack of activity. The system can only be restarted by pressing the reset button, at which point it resumes normal operation.

#include <Arduino.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include <Firebase_ESP_Client.h>
#include <addons/TokenHelper.h>
#include <BH1750.h>
#include <SimpleTimer.h>
#include "time.h"
#include "MHZ19.h"
#include <HTTPClient.h>
#include <Update.h>
#include <ESP32Encoder.h>
#include <ItemToggle.h>
#include <ItemSubMenu.h>
#include <ItemCommand.h>
#include <ItemInput.h>
#include <LcdMenu.h>
#include <OneButton.h>
#include <EEPROM.h>
// #include <esp_task_wdt.h>
// Provide the token generation process info.
#include "addons/TokenHelper.h"
// Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

// Firmware path and OTA
#define FIRMWARE_PATH "production/firmware.bin"
// #define FIRMWARE_PATH "test/firmware.bin"
const char *firmwareVersion = "1.0.5";

#define FIREBASE_USE_PSRAM

// Firebase Data Path

// String DeviceDocumentPath = "device1/";
// String DeviceDocumentPath = "device12/";

// Global variables
// TEMP SET POINTS

int tempSetPointOff;
int tempSetPointOn;

// HUM SET POINTS
int humdSetPointOff;
int humdSetPointOn;

// DEHUM SET POINTS
int dehumdSetPointOff;
int dehumdSetPointOn;

// gobal char Converter to the text

char tempOnValue[20];
char tempOffValue[20];
char dehumOnValue[20];
char dehumOffValue[20];
char humdOnValue[20];
char humdOffValue[20];

SimpleTimer timer;

const uint32_t connectTimeoutMs = 10000;

// Define EEPROM addresses for storing set point values
#define EEPROM_TEMP_SET_POINT_ON_ADDR 0
#define EEPROM_TEMP_SET_POINT_OFF_ADDR (EEPROM_TEMP_SET_POINT_ON_ADDR + sizeof(int))
[... 1304 lines cut ...]
{
  // Check if the value is non-zero and not null

  updateSetPoint("humd_set_point_off", "humd_set_point_off", value);
}

// BackLight ON/OFF
void toggleBacklight(uint16_t isOn)
{
  menu.setBacklight(isOn);
}

// Menu HideMenu

void hideMenu()
{
  // Perform actions to hide the menu

  Serial.println("Menu Hide");
  menu.hide(); // Clear the LCD screen or perform other actions as needed
  // delay(100);
  sensorsData = true;
}

Please check this code

mobizt, 2024-03-11:
You should follow the examples first and should read the comments in the example carefully for correctly usage.
https://github.com/mobizt/Firebase-ESP-Client/blob/main/examples/RTDB/Basic/Basic.ino

mobizt, 2024-03-11:
Please see the comments about `Firebase.ready()` in the examples.

closed 2024-03-11 by mobizt

Saipreetham0, 2024-03-11:
Okay Thank you

Saipreetham0, 2024-03-20:
Same issue and Unable to connect the wifi After ESP.reset();

mobizt, 2024-03-20:
Out of memory.

Saipreetham0, 2024-03-21:
okay any Solution for it ?

mobizt, 2024-03-21:
Nobody can solve it for you as it is your code and it is not pure Firebase library code but integration with third party libs.

You have to check and solve yourself.

Saipreetham0, 2024-03-21:
Can we reset the esp32 every 6 hour?

mobizt, 2024-03-21:
You should check free heap with `ESP.getFreeHrap()` periodically, if it continues to reduce, the memory leaks may happen.

Otherwise, you have to debug your code to be sure that `Firebase.ready()` was still executed in loop without blocking by your code.

Saipreetham0, 2024-03-22:
okay

Saipreetham0, 2024-03-22:
Find out if there is a solution to the out of memory issue, since more and more users are using Firebase

mobizt, 2024-03-22:
This is embedded device with usable  memory for code is about 200k.

You have to manage it and use it economically.

Saipreetham0, 2024-03-23:
okay bro