mobizt/Firebase-ESP-Client #654: WARN.mRunUntil: Terminating because the ssl engine closed.  > ERROR.write: Failed while waiting for the engine to enter BR_SSL_SENDAPP

manariamantion, 2024-03-14:
WARN.mRunUntil: Terminating because the ssl engine closed.  > ERROR.write: Failed while waiting for the engine to enter BR_SSL_SENDAPP

firebase connect one time, after some time letter, get this error and then not connected to firebase

mobizt, 2024-03-14:
If you want to post the issue here, you should provide the information about code, what you intend to do or what you expected and what is the result or error.

Just telling error of unknown action is not helpful for everyone to understand your context.

closed 2024-03-14 by mobizt

manariamantion, 2024-03-14:
//************** include library ***************
 #include <Preferences.h>        //for save state of pin in flash
 #include <SPI.h>
 #include <Wire.h>
 #include "RTClib.h" 
 #include <NTPClient.h>
 #include <Arduino.h>
 #include <WiFi.h>
 #include <Firebase_ESP_Client.h>
 #include <WiFiUdp.h>
 #include <Adafruit_GFX.h>        //ssd lib
 #include <Adafruit_SSD1306.h>    //ssd lib  
 #include "addons/TokenHelper.h" //Provide the token generation process info.
 #include "addons/RTDBHelper.h" //Provide the RTDB payload printing info and other helper functions.
 #include <WiFiMulti.h>
//################# Variables #################
 WiFiMulti wifiMulti;
 WiFiUDP ntpUDP;
 NTPClient timeClient(ntpUDP, "pool.ntp.org");
 Preferences preferences;
 RTC_DS1307 rtc;
 char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"}; 
 
 FirebaseData fbdo;  //2. Define FirebaseESP8266 data object for data sending and receiving 
 FirebaseAuth auth;
 FirebaseConfig config;
 bool signupOK = false; 
//################# define  ################# 
 #define API_KEY "AIzaSyCRxMrKqA4p3WeqOQqpYX56-L1XsEihBiA"     // Insert Firebase project API Key
 #define DATABASE_URL "pum-phouse-remot-consol.firebaseio.com" // Insert RTDB URLefine the RTDB URL 
 #define SCREEN_WIDTH 128 // OLED display width, in pixels
 #define SCREEN_HEIGHT 64 // OLED display height, in pixels
 #define OLED_RESET     -1 // Reset pin # (or -1 if sharing Arduino reset pin)
 #define SCREEN_ADDRESS 0x3C ///< See datasheet for Address; 0x3D for 128x64, 0x3C for 128x32
 Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET); 
 #define wifi_led 2
 #define motor 18
 #define light 19
 // #define motor2 13
 #define modem_power_pin 14
 #define ON 0 
 #define OFF 1
 const uint32_t connectTimeoutMs = 15000;
 int reconnect_time =  15000;
 int last_try_time = 0;  
 int prog_day = 5, prog_month = 3, prog_year = 24;
 int DAY, MONTH, YEAR, HH, MM, Sec, HHMM;
 int TLON, TLOF, TMON1, TMOF1, TMON2, TMOF2; //TIMER LIGHT ON,OFF, MOTOR1 ON,OFF, MOTOR2 ON,OFF
 int lightState1, motorState1, lightState2, motorState2;// 0 = manual, 1 = automatic, 1 = manual, 2 = auto
 int input_code ; // input code value from FB
 int MODE = 0;/* 0 = AUTO MODE, 1 = MANUAL WHEN WIFI TRUE*/
 int RTC_SYNC = 0; // 0 = rtc not sync after setup, 1 = rtc sync after setup,
 int UPDATEFLAGE;
//################### setup #################
 void setup() { // put your setup code here, to run once:  
  config.api_key = API_KEY; // Assign the api key (required) 
  config.database_url = DATABASE_URL; // Assign the RTDB URL (required) 
  //fbdo.setBSSLBufferSize(4096 /* Rx buffer size in bytes from 512 - 16384 */, 1024 /* Tx buffer size in bytes from 512 - 16384 */);
  PIN_SETUP();
  Wire.begin(); 
  Serial.begin(115200);   
  preferences.begin("my-app", false); 
  wifiMulti.addAP("r4jm3n4ri4", "r4jm3n4ri4");
  wifiMulti.addAP("", "");
  wifiMulti.addAP("JioFiber-amfda", "em3haichoo9xahPu");
  DATA_INIT();  
  if (!display.begin(SSD1306_SWITCHCAPVCC, SCREEN_ADDRESS)) { Serial.println(F("SSD1306 allocation failed"));  for(;;); }   
  display.display();   delay(1000);
  display.clearDisplay();  
  display.drawRect(0, 0, 128, 64, WHITE); display.display();
[... 182 lines cut ...]
      break;
     } 
   }
void input_code_0(void){ 
  Serial.println("DO NOTHING");  
 }
void input_code_1(void){    // WHEN INPUT CODE = 1 , MEANS MODE VALUE UPDATE 
 if (Firebase.RTDB.getInt(&fbdo, "/NEWDB/AA_MODE")) { if (fbdo.dataType() == "int") { MODE = fbdo.intData();}  } 
     Firebase.RTDB.setInt(&fbdo, "NEWDB/UPDATEFLAGE", 0) ? timerUpdateflageSettingtrue() : timerUpdateflageSettingfalse();   
 }
void input_code_2(void){    // WHEN INPUT CODE = 2 , MEANS TIMER VALUE UPDATE  
 if (Firebase.RTDB.getInt(&fbdo, "/NEWDB/A_TMON1")) { if (fbdo.dataType() == "int") { TMON1 = fbdo.intData();}  }
 if (Firebase.RTDB.getInt(&fbdo, "/NEWDB/A_TMOF1")) { if (fbdo.dataType() == "int") { TMOF1 = fbdo.intData();}  }
 if (Firebase.RTDB.getInt(&fbdo, "/NEWDB/A_TMON2")) { if (fbdo.dataType() == "int") { TMON2 = fbdo.intData();}  }
 if (Firebase.RTDB.getInt(&fbdo, "/NEWDB/A_TMOF2")) { if (fbdo.dataType() == "int") { TMOF2 = fbdo.intData();}  }
 if (Firebase.RTDB.getInt(&fbdo, "/NEWDB/A_TLON"))  { if (fbdo.dataType() == "int") { TLON  = fbdo.intData();}  }
 if (Firebase.RTDB.getInt(&fbdo, "/NEWDB/A_TLOF"))  { if (fbdo.dataType() == "int") { TLOF  = fbdo.intData();}  }
 preferences.putInt("TMON1", TMON1);
 preferences.putInt("TMOF1", TMOF1);
 preferences.putInt("TMON2", TMON2);  
 preferences.putInt("TMOF2", TMOF2);
 preferences.putInt("TLON",  TLON);        
 preferences.putInt("TLOF",  TLOF);
 Firebase.RTDB.setInt(&fbdo, "NEWDB/UPDATEFLAGE", 0) ? timerUpdateflageSettingtrue() : timerUpdateflageSettingfalse();   
 }