mobizt/Firebase-ESP-Client #656: Compilation error: 'RealtimeDatabase' does not name a type   #include <FirebaseClient.h>

manariamantion, 2024-03-15:
this is example code that provide with library, only cradential  change 
but way got this error, 
plz tell what i did wrong 

#include <Arduino.h>
#if defined(ESP32) || defined(ARDUINO_RASPBERRY_PI_PICO_W)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#elif __has_include(<WiFiNINA.h>)
#include <WiFiNINA.h>
#elif __has_include(<WiFi101.h>)
#include <WiFi101.h>
#elif __has_include(<WiFiS3.h>)
#include <WiFiS3.h>
#endif

#include <FirebaseClient.h>

#if defined(ESP8266) || defined(ESP32)
#include <WiFiClientSecure.h>
#endif

#define WIFI_SSID ""r4jm3n4ri4""
#define WIFI_PASSWORD ""r4jm3n4ri4""

// The API key can be obtained from Firebase console > Project Overview > Project settings.
#define API_KEY "XXXXXXXX"

// User Email and password that already registerd or added in your project.
#define USER_EMAIL "rajmenaria1985@gmail.com"
#define USER_PASSWORD "XXXXXXXX"
#define DATABASE_URL "XXXXXXXX"

void printError(int code, const String &msg);

void asyncCB(AsyncResult &aResult);

DefaultNetwork network; // initilize with boolean parameter to enable/disable network reconnection

UserAuth user_auth(API_KEY, USER_EMAIL, USER_PASSWORD);

FirebaseApp app;

WiFiClientSecure ssl_client;

// In case the keyword AsyncClient using in this example was ambigous and used by other library, you can change
// it with other name with keyword "using" or use the class name AsyncClientClass directly.

using AsyncClient = AsyncClientClass;

AsyncClient aClient(ssl_client, getNetwork(network));

RealtimeDatabase Database;

void setup()
{

    Serial.begin(115200);
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);

    Serial.print("Connecting to Wi-Fi");
    unsigned long ms = millis();
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(300);
    }
    Serial.println();
    Serial.print("Connected with IP: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    Serial.printf("Firebase Client v%s\n\n", FIREBASE_CLIENT_VERSION);

    Serial.println("Initializing app...");

    ssl_client.setInsecure();
#if defined(ESP8266)
    ssl_client.setBufferSizes(4096, 1024);
#endif

    app.setCallback(asyncCB);

    initializeApp(aClient, app, getAuth(user_auth));

    // Waits for app to be authenticated.
    // For asynchronous operation, this blocking wait can be ignored by calling app.loop() in loop().
    ms = millis();
    while (app.isInitialized() && !app.ready() && millis() - ms < 120 * 1000)
        ;

    app.getApp<RealtimeDatabase>(Database);

    Database.url(DATABASE_URL);

    Serial.println("[+] Synchronous Set... ");

    // Set int
    bool status = Database.set<int>(aClient, "/test/int", 12345);
    if (status)
        Serial.println("Set int is ok");
    else
        printError(aClient.lastError().code(), aClient.lastError().message());

    // Set bool
    status = Database.set<bool>(aClient, "/test/bool", true);
    if (status)
        Serial.println("Set bool is ok");
    else
        printError(aClient.lastError().code(), aClient.lastError().message());

    // Set string
    status = Database.set<String>(aClient, "/test/string", "hello");
    if (status)
        Serial.println("Set string is ok");
    else
        printError(aClient.lastError().code(), aClient.lastError().message());

    // Set json

    // Library does not provide JSON parser library, the following JSON writer class will be used with
    // object_t for simple demonstration.

    object_t json;
    JsonWriter writer;
    writer.create(json, "test/data", 123); // -> {"test":{"data":123}}
    // Or set the seialized JSON string to object_t as object_t("{\"test\":{\"data\":123}}")

    status = Database.set<object_t>(aClient, "/test/json", json);
    if (status)
        Serial.println("Set json is ok");
    else
        printError(aClient.lastError().code(), aClient.lastError().message());

    // Set array

    object_t arr;
    arr.initArray();                                                                                      // To use as Array placeholder
    writer.join(arr, 4, object_t(1), object_t(2), object_t(string_t("test")), object_t(boolean_t(true))); // -> [1,2,"test",true]
    // Or set the seialized JSON Array string to the object_t as object_t("[1,2,\"test\",true]")

    status = Database.set<object_t>(aClient, "/test/arr", arr);
    if (status)
        Serial.println("Set array is ok");
    else
        printError(aClient.lastError().code(), aClient.lastError().message());

    // Set float
    status = Database.set<number_t>(aClient, "/test/float", number_t(123.456, 2));
    if (status)
        Serial.println("Set float is ok");
    else
        printError(aClient.lastError().code(), aClient.lastError().message());

    // Set double
    status = Database.set<number_t>(aClient, "/test/double", number_t(1234.56789, 4));
    if (status)
        Serial.println("Set double is ok");
    else
        printError(aClient.lastError().code(), aClient.lastError().message());
}

void loop()
{
    // This function is required for handling async operations and maintaining the authentication tasks.
    app.loop();

    // (For Async Only) This required when different AsyncClients than used in FirebaseApp assigned to the Realtime database functions.
    Database.loop();
}

void asyncCB(AsyncResult &aResult)
{
    if (aResult.appEvent().code() > 0)
    {
        Serial.println("**************");
        Serial.printf("Event msg: %s, code: %d\n", aResult.appEvent().message().c_str(), aResult.appEvent().code());
    }

    if (aResult.isError())
    {
        Serial.println("**************");
        Serial.printf("Error msg: %s, code: %d\n", aResult.error().message().c_str(), aResult.error().code());
    }

    if (aResult.isDebug())
    {
        Serial.println("**************");
        Serial.printf("Debug msg: %s\n", aResult.debug().c_str());
    }
}

void printError(int code, const String &msg)
{
    Serial.printf("Error, msg: %s, code: %d\n", msg.c_str(), code);
}

manariamantion, 2024-03-15:
**### geting error at compile time**

Using board 'esp32' from platform in folder: release_v5.1-3662303f31/esp32/flags/cpp_flags" -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 "@C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/flags/defines" "-IC:\\Users\\GABBAR\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024215-7584-1e2q75c.kwzlh\\sketch_mar15a" -iprefix "C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/include/" "@C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/flags/includes" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/qio_qspi/include" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\cores\\esp32" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\variants\\esp32" "-C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\libraries\\WiFiClientSecure\\src" "@C:\\Users\\GABBAR\\AppData\\Local\\Temp\\arduino-sketch-EAAA463448EE51D7813708F4B899258D/build_opt.h" "@C:\\Users\\GABBAR\\AppData\\Local\\Temp\\arduino-sketch-EAAA463448EE51D7813708F4B899258D/file_opts" "c:\\Users\\GABBAR\\Documents\\Arduino\\libraries\\FirebaseClient-"@C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/flags/cpp_flags" -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 "@C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/flags/defines" "-IC:\\Users\\GABBAR\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024215-7584-1e2q75c.kwzlh\\sketch_mar15a" -iprefix "C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/include/" "@C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/flags/includes" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/qio_qspi/include" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\cores\\esp32" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\variants\\esp32" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\libraries\\WiFi\\src" "-Ic:\\Users\\GABBAR\\Documents\\Arduino\\libraries\\FirebaseClient-main\\src" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\libraries\\FS\\src" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\libraries\\Update\\src" "-IC:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\libraries\\WiFiClientSecure\\src" "@C:\\Users\\GABBAR\\AppData\\Local\\Temp\\arduino-sketch-EAAA463448EE51D7813708F4B899258D/build_opt.h" "@C:\\Users\\GABBAR\\AppData\\Local\\Temp\\arduino-sketch-EAAA463448EE51D7813708F4B899258D/file_opts" "C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\3.0.0-alpha3\\libraries\\Update\\src\\Updater.cpp" -o nul
"C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\esp-12.2.0_20230208/bin/xtensa-esp32-elf-g++" -c "@C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino-libs\\idf-release_v5.1-3662303f31/esp32/flags/cpp_flags" -w -Os -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"windows\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32:JTAGAdapter=default,PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none,EraseFlash=none,ZigbeeMode=default\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 "@C:\\Users\\GABBAR\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\esp32-arduino [... cut ...]
Using library FS at version 2.0.0 in folder: C:\Users\GABBAR\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\FS 
Using library Update at version 2.0.0 in folder: C:\Users\GABBAR\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\Update 
Using library WiFiClientSecure at version 2.0.0 in folder: C:\Users\GABBAR\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0-alpha3\libraries\WiFiClientSecure 
exit status 1

Compilation error: 'RealtimeDatabase' does not name a type

closed 2024-03-15 by manariamantion

mobizt, 2024-03-15:
The error was fixed. You can update the [FirebaseClient](https://github.com/mobizt/FirebaseClient) library with latest files.