mobizt/Firebase-ESP-Client #664: Question: How do handle token error code: -4

ateker01, 2024-04-12:
Intentionally I disable the internet access to the ESP8266 and receiving the following error as expected

00:13:20.046 -> Token info: type = id token (GITKit token), status = on request
00:13:25.066 -> Token info: type = id token (GITKit token), status = error

I eventually want it to quit trying so i can programmatically handle the error with the connectionError() method as seen below code. It however never quit trying - it continuously sends "on request" and receives "error"

What is the suggested way to handle the firebase connection failures?
 
  if(Firebase.RTDB.setJSON(&firebaseData, path, &jsonDefault)){
    }else{
      firebaseData.errorReason().c_str();
      connectionError();
      delay(100);
  }

Below is the setup and I am using version 4.4.14

  // Assign the api key (required)
  config.api_key = API_KEY;

  /* Assign the RTDB URL (required) */
  config.database_url = DATABASE_URL;
  
  // Assign the user sign in credentials
  auth.user.email = USER_EMAIL;
  auth.user.password = USER_PASSWORD;

  Firebase.reconnectWiFi(true);
  firebaseData.setResponseSize(4096);

  // Assign the callback function for the long running token generation task
  config.token_status_callback = tokenStatusCallback; //see addons/TokenHelper.h

  // Assign the maximum retry of token generation
  config.max_token_generation_retry = 5;

  // Initialize the library with the Firebase authen and config
  Firebase.begin(&config, &auth);

  delay(300);

mobizt, 2024-04-12:
The code you post above is generally can be found from library examples and it does not help anything.

You should start at the code that works as from examples and try to adapt from it.

Something that you changed in examples that is the cause of issue that it is error in coding or logic.

Anyway, delay should be avoided, you don't have any hardware event to be waiting.

closed 2024-04-12 by mobizt

mobizt, 2024-04-12:
If you use WiFiManager, you have to notice this.
https://github.com/mobizt/Firebase-ESP-Client/blob/02c7bd1cc95e14a5efc22b9ab9e8037643f4e24e/examples/RTDB/Basic/Basic.ino#L118-L119