mobizt/Firebase-ESP-Client #674: firebase esp32 client library clashes with esp mail client library

alabadan01, 2024-08-09:
i am doing a fingerprint attendance system using real-time firebase as the databse and also sending the attendance sheet after every lecture to an email acccount for later purposes and reasons. However, there is a compilation issue at the end of the journey. further investigations shows that there is a conflict between the both libraries being used. something indicating that 
[image]
Both libraries are defining the same BSSL_TCP_Client class and its methods, causing multiple definition errors.
The error messages indicate that functions from BSSL_TCP_Client are defined in both libraries, leading to conflicts during linking.

**IDE and its version:**
 - Arduino,
 - Version [2.3.2]

mobizt, 2024-08-09:
This library is end of life.

mobizt, 2024-08-09:
https://github.com/mobizt/Firebase-ESP-Client?tab=readme-ov-file#we-have-moved-to-the-new-library

closed 2024-08-09 by mobizt

mobizt, 2024-08-09:
Both libraries have the folder `src/client/SSLClient`, which you have to remove one which depending on which library was already included it.

if `ESP_Mail_Client.h` was included after `FirebaseXXXXX.h`, the folder [src/client/SSLClient](https://github.com/mobizt/ESP-Mail-Client/tree/master/src/client/SSLClient) in the `ESP-Mail-Client` library installation folder should be removed.

alabadan01, 2024-08-09:
using the new library and its the same. the evidence of this installation is in the screenshot showing its been installed
[image]
[image]

[image]

mobizt, 2024-08-09:
Read my post above again.

mobizt, 2024-08-09:
You have to uninstall all Firebase libraries and install only one and follow my comment above for redundant folders deletion.

alabadan01, 2024-08-09:
ive got new errors

mobizt, 2024-08-09:
Nothing is error if you follow my comment correctly.

alabadan01, 2024-08-10:
do i use the firebase client library now?

alabadan01, 2024-08-10:
the conflict is now with the firebse json library and the firebase esp32 library

mobizt, 2024-08-10:
https://github.com/mobizt/Firebase-ESP32?tab=readme-ov-file#we-have-moved-to-the-new-library

mobizt, 2024-08-10:
If you want to use new library, you have to read the documentation thoroughly and follow the library examples.