mobizt/Firebase-ESP-Client #677: Firebase Library conflicting

chamikasahan, 2024-09-20:
Hi sir, I have used 2 seperate codes for check my gps and mic. but i used firebase esp client library for gps and  firebase client library for INMP441 mic. 

yesterday when i try to merge these two code, libraries are conflicting. i have tried more than 10 hours to fix this issue. there are no errors in the code. but after compile, it show "exit status"  and above of it, there are lot of errors which i dont have my code. i think those errors in the library files.   

can You help me sir to fix this error. i wanted to use new firebase client library for my gps code. i used set example code in the library provided. but Im stuck on that.  

below is my full code that i merge together sir. please could you please look it and give me a support. this is very urgent help sir. Im kindly request you to help me 🙏

`#include <Arduino.h>
#include <TinyGPS++.h>
#include <driver/i2s.h>
#include <SPIFFS.h>
#include <WiFi.h>
#include <Firebase_ESP_Client.h>
#include <FirebaseClient.h>
#include <WiFiClientSecure.h>

// Provide the token generation process info.
#include "addons/TokenHelper.h"
// Provide the RTDB payload printing info and other helper functions.
#include "addons/RTDBHelper.h"

// Firebase and Wi-Fi Config
#define WIFI_SSID "Chamika"
#define WIFI_PASSWORD "24682468"
#define API_KEY "AIzaSyAIQv6xLIrPtM9X2B6K6uoFJuP7JP8NhuU"
#define DATABASE_URL "guardianalert-ddde5-default-rtdb.firebaseio.com"
#define STORAGE_BUCKET_ID "guardianalert-ddde5.appspot.com"

// I2S Configuration for recording
#define I2S_WS 25
#define I2S_SD 33
#define I2S_SCK 32
#define I2S_PORT I2S_NUM_0
#define I2S_SAMPLE_RATE 16000
#define I2S_SAMPLE_BITS 16
#define I2S_READ_LEN 16 * 1024
#define RECORD_TIME 20  // Seconds
#define I2S_CHANNEL_NUM 1
#define FLASH_RECORD_SIZE I2S_CHANNEL_NUM * I2S_SAMPLE_RATE * I2S_SAMPLE_BITS / 8 * RECORD_TIME
const char filename[] = "/recording.raw";  // Changed to .raw
File file;
void fileCallback(File &file, const char *filename, file_operating_mode mode);
// Define push button pin
#define BUTTON_PIN 23

// Define file config for uploading
FileConfig media_file("/recording.raw", fileCallback);

// HardwareSerial for GPS and GPS object
HardwareSerial GPS_Serial(2); // (Rx, Tx)
TinyGPSPlus gps;

using AsyncClient = AsyncClientClass;
// Firebase objects
WiFiClientSecure ssl_client;
DefaultNetwork network;
FirebaseData fbdo;
FirebaseAuth auth;
FirebaseConfig config;
Storage storage;
// Tracking the button state
unsigned long sendDataPrevMillis = 0;
bool signupOK = false;
int buttonState = 0;
bool taskCompleted = false;

FirebaseApp app;

AsyncClient aClient(ssl_client, getNetwork(network));

// Function declarations
void SPIFFSInit();
void i2sInit();
void i2s_adc(void* arg);
void uploadToFirebase();
bool fileExists(const char* path);

void setup() {
  Serial.begin(115200);
  GPS_Serial.begin(9600, SERIAL_8N1, 16, 17);  // Use pins 16 and 17 for RX and TX

  // Set up the push button pin
  pinMode(BUTTON_PIN, INPUT_PULLUP);
 
  // Initialize Wi-Fi
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.print("Connecting to Wi-Fi");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(300);
  }
  Serial.println();
  Serial.print("Connected with IP: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  // Firebase configuration
  config.api_key = API_KEY;
  config.database_url = DATABASE_URL;

  // Sign up for Firebase
  if (Firebase.signUp(&config, &auth, "", "")) {
    Serial.println("Firebase signup successful");
    signupOK = true;
  } else {
    Serial.printf("Firebase signup failed: %s\n", config.signer.signupError.message.c_str());
  }

  // Initialize SPIFFS and I2S
  SPIFFSInit();
  i2sInit();

  // Token status callback
  config.token_status_callback = tokenStatusCallback;
  Firebase.begin(&config, &auth);
  Firebase.reconnectWiFi(true);
}

void loop() {
  // Handle GPS communication
  smartDelay(1000);

  if (Firebase.ready() && signupOK && (millis() - sendDataPrevMillis > 5000 || sendDataPrevMillis == 0)) {
    sendDataPrevMillis = millis();

    if (gps.location.isValid()) {
      double latitude = gps.location.lat();
      double longitude = gps.location.lng();
      double altitude = gps.altitude.meters();

      // Send GPS data to Firebase
      Firebase.RTDB.setFloat(&fbdo, "/gps/latitude", latitude);
      Firebase.RTDB.setFloat(&fbdo, "/gps/longitude", longitude);
      Firebase.RTDB.setFloat(&fbdo, "/gps/altitude", altitude);
    }
  }

  // Handle push button press
  if (Firebase.ready() && signupOK) {
    buttonState = digitalRead(BUTTON_PIN);

    if (buttonState == LOW) { // Button pressed
      if (Firebase.RTDB.setInt(&fbdo, "/alertStatus", 1)) {
        Serial.println("Button pressed! Alert sent to Firebase.");
       
        // Start recording audio and upload
        xTaskCreate(i2s_adc, "i2s_adc", 1024 * 4, NULL, 1, NULL);  // Start I2S ADC task
      }
    } else {
      Firebase.RTDB.setInt(&fbdo, "/alertStatus", 0);
    }
   
    delay(500);
  }
}

static void smartDelay(unsigned long ms) {
  unsigned long start = millis();
  do {
    while (GPS_Serial.available()) {
      gps.encode(GPS_Serial.read());
    }
  } while (millis() - start < ms);
}

// Initialize SPIFFS for audio recording storage
void SPIFFSInit() {
  if (!SPIFFS.begin(true)) {
    Serial.println("SPIFFS initialization failed!");
    while (1) yield();
  }

  if (SPIFFS.exists(filename)) {
    SPIFFS.remove(filename);
  }

  file = SPIFFS.open(filename, FILE_WRITE);
  if (!file) {
    Serial.println("File creation failed!");
    while (1) yield();
  }
}

// Initialize I2S for audio recording
void i2sInit() {
  i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX),
    .sample_rate = I2S_SAMPLE_RATE,
    .bits_per_sample = i2s_bits_per_sample_t(I2S_SAMPLE_BITS),
    .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
    .communication_format = i2s_comm_format_t(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB),
    .intr_alloc_flags = 0,
    .dma_buf_count = 16,
    .dma_buf_len = 1024,
    .use_apll = 1
  };
  i2s_driver_install(I2S_PORT, &i2s_config, 0, NULL);

  const i2s_pin_config_t pin_config = {
    .bck_io_num = I2S_SCK,
    .ws_io_num = I2S_WS,
    .data_out_num = -1,
    .data_in_num = I2S_SD
  };
  i2s_set_pin(I2S_PORT, &pin_config);
}

// Record audio and store it in SPIFFS
void i2s_adc(void* arg) {
  int flash_wr_size = 0;
  size_t bytes_read;
  char* i2s_read_buff = (char*)malloc(I2S_READ_LEN);
 
  if (!i2s_read_buff) {
    Serial.println("Memory allocation failed!");
    vTaskDelete(NULL);
  }

  Serial.println("Recording started...");
  while (flash_wr_size < FLASH_RECORD_SIZE) {
    i2s_read(I2S_PORT, (void*)i2s_read_buff, I2S_READ_LEN, &bytes_read, portMAX_DELAY);
    if (bytes_read > 0) {
      file.write((const uint8_t*)i2s_read_buff, bytes_read);
      flash_wr_size += bytes_read;
    }
  }

  file.close();
  free(i2s_read_buff);
  uploadToFirebase();  // Upload to Firebase after recording

  vTaskDelete(NULL);
}

// Upload recorded file to Firebase
void uploadToFirebase() {
  if (fileExists(filename) && !taskCompleted) {
    taskCompleted = true;
    // Upload audio file to Firebase Storage
     storage.upload(
      aClient,
      FirebaseStorage::Parent(STORAGE_BUCKET_ID, "audio/recording2.raw"),
      getFile(media_file),
      "audio/raw",
      asyncCB,
      "uploadTask");
  }
}

// Check if file exists
bool fileExists(const char* path) {
  File file = SPIFFS.open(path);
  if (!file || file.isDirectory()) {
    return false;
  }
  file.close();
  return true;
}

// Async callback function for Firebase upload
void asyncCB(AsyncResult& aResult) {
  if (aResult.uploadProgress() && aResult.uploadInfo().total == aResult.uploadInfo().uploaded) {
    Serial.print("Download URL: ");
    Serial.println(aResult.uploadInfo().downloadUrl);
  }
}
 `

mobizt, 2024-09-20:
You should use only one Firebase library.

closed 2024-09-20 by mobizt

chamikasahan, 2024-09-20:
Can you please help me to that sir 🥺 i tried to move firebase esp library. but then mic code not working. when i try to move firebaseclient library, GPS  not correctly updating my firebase sir. Im a beginner for this codes sir

mobizt, 2024-09-20:
Remove (uninstall) all Firebase libraries and reinstall only one library.

mobizt, 2024-09-21:
You cannot merge the code from two Firebase libraries because it contains the same local built-in SSL client called `ESP_SSLClient`, which cause compilation error otherwise you have to delete built-in SSL client from another including Firebase library.

For `FirebaseClient` , you have to read its documentation thoroughly and follow the library examples to ensure you understand the concept and usage.

The `FirebaseClient` library is async library by design but it can use in sync operation too.

When using `FirebaseClient` library in async mode, data will be sent and received to/from the server as small chunks with sent/receive timeout limit and these operations were done in the main loop.  If your code contains delay or some blocking code in the main loop, it will cause sent/receive timed out and the async operations will fail.

You should not do some blocking tasks (code) when the async task is currently running. You should check that there are no async tasks in async client queue when you want to do some blocking tasks e.g. reading/writing files, reading GPS data.

You can use `FirebaseClient` in sync operation, but I still not recommend you use delay to just halt the processor from running the program. The delay is suitable and acceptable for waiting another connected device data or initialization.

In the `Firebase-ESP-Client` library, you can check the library examples folder for the code examples. You can use library to read/write Realtime database and upload/download data from Firebase storage with less delay restriction.

chamikasahan, 2024-09-21:
using firebase_esp_client library, i have modified the code. now its working !!